mavlink/rust-mavlink #10: Is this repository still maintained?

podhrmic, 2018-12-14:
Hi, just wondering if this repo is still maintained - I noticed a couple of pending PRs.
Thanks for the info.

podhrmic, 2018-12-17:
@hamishwillee thanks for the idea - I was playing with Dronecode SDK in the past. rust-mavlink was more interesting for me, because it is easier to integrate into other rust projects (like [pixhawk-to-UxAS-proxy](https://github.com/GaloisInc/pixhawk-proxy) ). So either way, having a stand-alone Rust library that understands Mavlink would be a good start.

Maybe moving this repo under [mavlink](https://github.com/mavlink) would be a better way?

hamishwillee, 2018-12-17:
@podhrmic I'm not opposed, but we need to get 3DR to release it, and Dronecode/MAVlink to accept it. MAVlink is pretty much a volunteer organisation anyway, so this wouldn't necessarily result in more resource. But at least there would be a team to accept changes.

mrpollo, 2018-12-18:
I don't own this repo either, and I have no special access to 3DR, but @kevinmehall still works there and is the owner of this repo.

This is a cool utility but doesn't provide a full mavlink API, if anyone is looking for this I would suggest discussing with the DronecodeSDK team, I'm sure a Rust wrapper would be welcomed!

As far as moving it to the mavlink org, we would need to bring it up on a dev call, but unless someone offers to maintain and triage my vote would be against.

tstellanova, 2019-01-25:
@mrpollo One quandary this apparently abandoned repo poses to the rust development community, from my perspective, is that it is the source of the "mavlink" named crate hosted on crates.io . @mrpollo @tcr3dr or @3drobotics would you be willing to add another owner/maintainer to the crate?  My suggestion is that if 3DR is unable to maintain this repo or release it, then a new maintainer would need to swap a new implementation onto crates.io. Ideally they wouldn't need to create a new crate with a name like "mavlink-maintained".

podhrmic, 2019-01-25:
Good point @tstellanova 
Just an FYI, I am happy to contribute/help maintain this going forward. Also if you are interested, I refactored the code generation, now it uses quote! macros and includes Mav enums and better ser/deser functions: https://github.com/GaloisInc/rust-mavlink/tree/refactor

tstellanova, 2019-01-27:
Great progress Michal!  I've forked the Galois repo and started to integrate
[fixes for TCP and further support for mavlink2](https://github.com/tstellanova/rust-mavlink/tree/fix_tcp_recv).
Once this is stable I'll publish a new MAVLink crate to crates.io and add you as an owner if you like. (@mrpollo et al: I could alternatively publish to the existing crate, but you'd need to add me as a crate owner)

kevinmehall, 2019-01-27:
3DR doesn't have anyone actively responsible for this repository anymore. I can put a little time into code review, but I think it makes sense to add additional maintainers and ideally transfer it somewhere like the `mavlink` org.

I reviewed @podhrmic's pending [serial PR](https://github.com/3drobotics/rust-mavlink/pull/9), and the updates in both of your repos looks like a good start as well. If you open a PR I can post my feedback there.

I have `cargo publish` access via the 3dr-engineering team, but GitHub teams don't have access to add new owners on crates.io. @mrpollo or @tcr, would you mind adding `kevinmehall`, `podhrmic` and `tstellanova` at https://crates.io/crates/mavlink/owners? I'll see about getting you commit access to this repo as well.

podhrmic, 2019-01-28:
@tstellanova cool! I am happy you found my fork useful:-)

@kevinmehall I will fix the PR accordingly, and later open a more comprehensive PR my other changes.

mrpollo, 2019-01-28:
Interesting, I didn't know I had access to that, I just gave access to @kevinmehall @tstellanova @podhrmic and @hamishwillee

podhrmic, 2019-01-28:
Thanks @mrpollo !

kevinmehall, 2019-04-11:
Repo transferred to `mavlink` and I've added @podhrmic  and @tstellanova as collaborators :tada:

Thanks for your help @mrpollo and @dagar!

closed 2019-04-11 by kevinmehall