mavlink/rust-mavlink #14: Getting ready for an official release

podhrmic, 2019-05-28:
This is a tracking issue to make rust-mavlink an officially supported Mavlink implementation.

What needs to happen?

- [ ] Split messages into one file per message (similar to what mavgen does)
- [ ] Add quickcheck tests for each message
- [ ] Add fuzz testing
- [ ] change enums to uints
- [ ] (optional) Build actual enums on top of message enums
- [ ] Mavlink 2 support, including signing
- [ ] Quote update
- [ ] ?

podhrmic, 2019-06-05:
FYI work in progress, check out https://github.com/GaloisInc/rust-mavlink/tree/fuzz_testing for the latest commits

podhrmic, 2019-06-12:
Added fuzz testing, and started adding quickcheck tests.

Next steps:
- figure out how to implement `Arbitrary` for `ArrayVec`
- implement `Arbitrary` for `MavMessage` and `MavFrame` to complete quickcheck tests

@hamishwillee do you think we need to have hardcoded test vectors as well? Or is a random (quickcheck) testing considered sufficient? Meaning - I generate a message (let's say `HEARTBEAT`) with random values and try to serialize and then deserialize it?

hamishwillee, 2019-06-17:
@podhrmic Sorry for the delay, I'm travelling.

> do you think we need to have hardcoded test vectors as well? Or is a random (quickcheck) testing considered sufficient? Meaning - I generate a message (let's say HEARTBEAT) with random values and try to serialize and then deserialize it?

Short version ... no.

Generally the steer from @peterbarker on the mavgen library has been "try to test to the same level as the other libraries" - ie C. I believe the C library tests every message but I don't know what values are passed in - ie whether they are random or not. 
My gut feeling is that random is good, but we may need to tune this to avoid false positives for some messages. 

What we _should_ do is work out a minimum test strategy for any generator library and document it - so you don't have to ask these questions. Also once we know what is tested, do a gap analysis to see if the coverage is reasonable. @peterbarker Do you have a summary anywhere of what tests are run for pymavlink and the C libraries?

podhrmic, 2019-06-17:
Thanks @hamishwillee for the update. 

FYI updates on my side if you are interested:
- switching to use `ArrayVec` only for arrays > 32 elements, otherwise will use slices
- working on generating enums as defined in the XML (for easier typechecking) that allow unknown values

podhrmic, 2020-03-11:
Closing as this is not relevant anymore. Some of the ideas (fuzz testing, etc) can be incorporated later if needed.

closed 2020-03-11 by podhrmic