mavlink/rust-mavlink #24: Common message set (common.xml) is missing messages

michael-p, 2020-01-03:
Dear all,

it seems that the message definition file (`common.xml`) that is used to generate the code in this crate is pretty old, and misses quite a few messages. In my case I stumbled across this because of the `LOGGING_DATA` and `LOGGING_DATA_ACKED` messages (both are Mavlink 2 messages).

It would be great if this crate could be updated to the most recent version!

Thanks
Michael

podhrmic, 2020-01-21:
Can you please try again with the latest master? #27 addressed this.

michael-p, 2020-01-22:
I just tested this, but sadly the logging related messages seem to not work. I'm currently using my [own fork](https://github.com/michael-p/rust-mavlink) where I simply added the messages I need to the common.xml file and the mavlink log streaming works fine with that. However, just switching to this repo's master branch results in the logging request being rejected by the flight controller (PX4 1.10).

Also, I noted another problem with the `VISION_POSITION_ESTIMATE` message which gained two new fields (reset_counter and covariance) - I just set `reset_counter` to 0 and covariance to `[NAN; 21]` but then, PX4 ignores the message completely.

I will have more time to debug this tomorrow but just wanted to give you a quick feedback in case you were planning on releasing a new version to crates.io - would be great if you could delay that a bit :) I will get back to you with more information in the next few days.

michael-p, 2020-01-23:
I did some more test and it looks like message with `<extensions/>` are not serialized and deserialized correctly (or, at least, it seems to be incompatible with PX4).

I tested with `VISION_POSITION_ESTIMATE` (send from Rust to PX4, that is, serialization) and `COMMAND_ACK` and `ATTITUDE_QUATERNION` (deserialization). In both cases, messages are not received at the other end, however, when I comment out the `<extensions/>` in `common.xml` everything works as expected.

The following example shows the behavior for deserialization and can be tested with [PX4 Gazebo SITL](https://dev.px4.io/master/en/simulation/gazebo.html). The expected behavior is that the program connects to the simulated MAV and basically immediately exits with a success message because it received the `ATTITUDE_QUATERNION` message. With the updated `common.xml`, however, it never receives that message.

```
use mavlink::{common::*, MavConnection};
use std::{sync::Arc, thread, time::Duration};

fn main() {
    let vehicle: Arc<dyn MavConnection + Send + Sync> = mavlink::connect("udpin:0.0.0.0:14540")
        .expect("connect to MAV")
        .into();
    println!("connected!");
    thread::spawn({
        let vehicle = Arc::clone(&vehicle);
        move || loop {
            vehicle
                .send_default(&heartbeat_message())
                .expect("send heartbeat");
            thread::sleep(Duration::from_millis(500));
        }
    });
    loop {
        let (_, message) = vehicle.recv().expect("receive message");
        if let MavMessage::ATTITUDE_QUATERNION(_) = message {
            println!("Got attitude quternion! SUCCESS!!");
            break;
        }
    }
}

fn heartbeat_message() -> MavMessage {
    MavMessage::HEARTBEAT(HEARTBEAT_DATA {
        custom_mode: 0,
        mavtype: MavType::MAV_TYPE_QUADROTOR,
        autopilot: MavAutopilot::MAV_AUTOPILOT_GENERIC,
        base_mode: MavModeFlag::empty(),
        system_status: MavState::MAV_STATE_STANDBY,
        mavlink_version: 0x3,
    })
}
```

patrickelectric, 2020-01-23:
@michael-p I confirm that extensions are being compiled, will try to investigate it.

patrickelectric, 2020-01-23:
@michael-p please check #30, it should fix your problem

michael-p, 2020-01-24:
Thanks Patrick, I checked your PR, and with the new `emit-extensions` feature disabled everything works as expected.

Although this does fix my use-case I'm still not sure there might not be a deeper cause for these problems. As far as I understand the [Mavlink documentation about extensions](https://mavlink.io/en/guide/define_xml_element.html#message_extensions), these extension fields should just be ignored by an implementation that does not implement them. So even if PX4 really had these extensions disabled, everything should still just work and not completely drop messages (in fact, looking at the code I'm quite sure px4 actually has the extensions enabled in their Mavlink interface).

In other words, couldn't it be that there is a problem in the way this library handles (serializes/deserialize) extensions? Did you successfully test this with another Mavlink implementation? I'm no Mavlink expert, just a user, so please take everything I say with a grain of salt :)

patrickelectric, 2020-01-24:
@michael-p I never used the extensions feature myself, but from the binary serialization/deserialization code, rust-mavlink is doing it in correctly and without problems. 
We should investigate it a bit further, but the extension fields as features already help us to debug and one step further to provide the compatibility with all protocol definition files.

michael-p, 2020-01-24:
You are right, having this behind a feature flag is definitively a way to move forward.
However, I think we should at least put a comment in the `Cargo.toml` next to that feature explaining that there might still be issues with extensions, maybe even point to this issue. Otherwise, I imagine this could be quite frustrating for a new user who discovers and enables the `emit-extensions` feature, and suddenly certain messages just do not work anymore. Again, it's not just the extension *fields* that are not working, the whole *message* is dropped completely as soon as the feature is enabled!

PR #31 (2020-01-29): Use mavlink submodule and remove xml file
Fix #24
Fix #27  
Fix #29
closed 2020-01-29 by podhrmic