mavlink/rust-mavlink #33: A new release before multiple messages definitions

patrickelectric, 2020-01-30:
Right now we have the full common message definition available.
It's a good time to do a new release before the multiple message definition integration [ardupilotmega, autoquad, minimal, standard, ...].

patrickelectric, 2020-01-30:
ping @podhrmic

podhrmic, 2020-01-31:
I can push to crates.io (I *think* I have access)
@patrickelectric do you want to make any additional changes in the documentation/etc. before pushing the new release?

patrickelectric, 2020-01-31:
@podhrmic I did some small improvements and updates in Cargo to 0.7 release,
you can check it here: https://github.com/mavlink/rust-mavlink/pull/34

podhrmic, 2020-01-31:
@patrickelectric I don't have access rights to the [mavlink crate](https://crates.io/crates/mavlink) 

@tstellanova @mrpollo @kevinmehall @tcr3dr can you please add me as an owner? (or publish the new version yourself)

patrickelectric, 2020-01-31:
@podhrmic We could also do such thing via travis, if someone could setup the token.

tstellanova, 2020-02-01:
Sent you an invite Michal— let me know if I need to do more.

podhrmic, 2020-02-01:
Done! https://crates.io/crates/mavlink

closed 2020-02-01 by podhrmic

patrickelectric, 2020-02-01:
[image]