mavlink/rust-mavlink #35: Release 0.7 requires recompilation on every build

michael-p, 2020-02-01:
Dear all,

I just noticed that the recently published version 0.7 is rebuild by Cargo on every run, even though nothing has been modified.

To reproduce, create a new project and add mavlink to it:
```
$ cargo new --bin mavlink-test
$ cd mavlink-test
$ cargo add mavlink
$ cargo build
```
Then, just run the project twice without modifying anything in between:
```
$ cargo run && cargo run
   Compiling mavlink v0.7.0
   Compiling mavlink-test v0.1.0 (/Users/michaelp/Desktop/mavlink-test)
    Finished dev [unoptimized + debuginfo] target(s) in 17.18s
     Running `target/debug/mavlink-test`
Hello, world!
   Compiling mavlink v0.7.0
   Compiling mavlink-test v0.1.0 (/Users/michaelp/Desktop/mavlink-test)
    Finished dev [unoptimized + debuginfo] target(s) in 17.44s
     Running `target/debug/mavlink-test`
Hello, world!
```
Notice that both the mavlink crate as well as the application itself were compiled again.

For comparision, this is the output with version 0.6:
```
$ cargo run && cargo run
   Compiling mavlink-test v0.1.0 (/Users/michaelp/Desktop/mavlink-test)
    Finished dev [unoptimized + debuginfo] target(s) in 0.25s
     Running `target/debug/mavlink-test`
Hello, world!
    Finished dev [unoptimized + debuginfo] target(s) in 0.05s
     Running `target/debug/mavlink-test`
Hello, world
```

I'm using this crate in a fairly large application where compile times are measured in minutes, so having to rebuild even though nothing has changed is pretty annoying as you can probably imagine :)

Thanks,
Michael

podhrmic, 2020-03-02:
Hmm, I wonder if this is because we now [checkout](https://github.com/mavlink/rust-mavlink/blob/master/build/main.rs#L19) `common.xml` from mavlink submodule, rather than using [a static copy of the file](https://github.com/mavlink/rust-mavlink/blob/c24eed107cbd8de083948a89c19dc9246dfe18ab/build/main.rs).

We should pin to a specific mavlink commit anyway, so `git checkout ...` would be better. I will see if pinning a specific commit makes this issue go away.

podhrmic, 2020-03-12:
@michael-p this issue is currently fixed in `master`, and will be shortly published as version `0.8` on crates.io

podhrmic, 2020-03-16:
@michael-p we just released `0.8` - I recommend updating your `Cargo.toml` as this will fix your issue.

closed 2020-03-16 by podhrmic

michael-p, 2020-03-16:
Thank you Michal!