mavlink/rust-mavlink #37: MavMessage::parse panics on invalid message

laptou, 2020-02-19:
https://github.com/mavlink/rust-mavlink/blob/8c43fe7fa2128eaa001c6640fafc746e01a7afd3/build/parser.rs#L195-L202

`MavMessage::parse` is set up to `unwrap()` the value that is provided by `serde`.

Why can't it just return `None` instead of unwrapping? This way, invalid data will not lead to a panic.

patrickelectric, 2020-03-02:
Is this still a valid issue ?

laptou, 2020-03-02:
I generally think that methods in a library should avoid panicking whenever possible (and here, it's definitely possible to avoid panicking by just returning `None`/`Err`). However, it's no longer actively causing problems in my application, so I'm not going to push on it.

laptou, 2020-04-15:
I have a MAVProxy server that is sending `COMMAND_ACK` messages with a `type` value of 11. I don't know what that is, but it seems like these messages can safely be ignored, so I would like the option to ignore them. However, the current behaviour of this crate is to panic when you try to parse an unknown enum value (the result of `unwrap`ing the result of using `FromPrimitive` in parser.rs).

PR #49 (2020-05-02): Return Results instead of panicking
Resolves #37. All of the functions that involve parsing a `MavMessage` will now return a `Result<Message, MessageReadError>` instead of panicking upon encountering invalid data.
closed 2020-05-02 by podhrmic