mavlink/rust-mavlink #58: feature ardupilotmega fail to build

patrickelectric, 2020-06-19:
```
error[E0432]: unresolved imports `crate::icarous`, `crate::uavionix`
 --> /home/patrick/git/blue/rust-mavlink/target/debug/build/mavlink-0dcff0587dd9cbda/out/ardupilotmega.rs:2:24
  |
2 | use crate::{common::*, icarous::*, uavionix::*};
  |                        ^^^^^^^     ^^^^^^^^ maybe a missing crate `uavionix`?
  |                        |
  |                        maybe a missing crate `icarous`?

error[E0433]: failed to resolve: maybe a missing crate `uavionix`?
    --> /home/patrick/git/blue/rust-mavlink/target/debug/build/mavlink-0dcff0587dd9cbda/out/ardupilotmega.rs:3531:21
     |
3531 |     uavionix(crate::uavionix::MavMessage),
     |                     ^^^^^^^^ maybe a missing crate `uavionix`?

error[E0433]: failed to resolve: maybe a missing crate `icarous`?
    --> /home/patrick/git/blue/rust-mavlink/target/debug/build/mavlink-0dcff0587dd9cbda/out/ardupilotmega.rs:3532:20
     |
3532 |     icarous(crate::icarous::MavMessage),
     |                    ^^^^^^^ maybe a missing crate `icarous`?

error[E0433]: failed to resolve: maybe a missing crate `uavionix`?
    --> /home/patrick/git/blue/rust-mavlink/target/debug/build/mavlink-0dcff0587dd9cbda/out/ardupilotmega.rs:3644:41
     |
3644 |                 if let Ok(msg) = crate::uavionix::MavMessage::parse(version, id, payload) {
     |                                         ^^^^^^^^ maybe a missing crate `uavionix`?

error[E0433]: failed to resolve: maybe a missing crate `icarous`?
    --> /home/patrick/git/blue/rust-mavlink/target/debug/build/mavlink-0dcff0587dd9cbda/out/ardupilotmega.rs:3647:41
     |
3647 |                 if let Ok(msg) = crate::icarous::MavMessage::parse(version, id, payload) {
     |                                         ^^^^^^^ maybe a missing crate `icarous`?

error[E0433]: failed to resolve: maybe a missing crate `uavionix`?
    --> /home/patrick/git/blue/rust-mavlink/target/debug/build/mavlink-0dcff0587dd9cbda/out/ardupilotmega.rs:3788:30
     |
3788 |                 match crate::uavionix::MavMessage::message_id_from_name(name) {
     |                              ^^^^^^^^ maybe a missing crate `uavionix`?

error[E0433]: failed to resolve: maybe a missing crate `icarous`?
    --> /home/patrick/git/blue/rust-mavlink/target/debug/build/mavlink-0dcff0587dd9cbda/out/ardupilotmega.rs:3792:30
     |
3792 |                 match crate::icarous::MavMessage::message_id_from_name(name) {
     |                              ^^^^^^^ maybe a missing crate `icarous`?

error[E0433]: failed to resolve: maybe a missing crate `uavionix`?
    --> /home/patrick/git/blue/rust-mavlink/target/debug/build/mavlink-0dcff0587dd9cbda/out/ardupilotmega.rs:4041:30
     |
4041 |                 match crate::uavionix::MavMessage::extra_crc(id) {
     |                              ^^^^^^^^ maybe a missing crate `uavionix`?

error[E0433]: failed to resolve: maybe a missing crate `icarous`?
    --> /home/patrick/git/blue/rust-mavlink/target/debug/build/mavlink-0dcff0587dd9cbda/out/ardupilotmega.rs:4045:30
     |
4045 |                 match crate::icarous::MavMessage::extra_crc(id) {
     |                              ^^^^^^^ maybe a missing crate `icarous`?

error: aborting due to 9 previous errors

Some errors have detailed explanations: E0432, E0433.
For more information about an error, try `rustc --explain E0432`.
error: could not compile `mavlink`.

To learn more, run the command again with --verbose.
 ✘  ~/git/blue/rust-mavlink   master  

```

laptou, 2020-06-19:
Are the `icarous` and `uavionix` features enabled?

patrickelectric, 2020-06-19:
Thanks @laptou, I was able to figure out it, there is an open PR for that: #60

PR #60 (2020-06-22): Cargo: ardupilot needs common, icarous and uavionix as described in the xml

Fix #58
commit 3d7dbde7 (2020-06-22): Cargo: Fix features dependency (#60)

Fix #58
closed 2020-06-22 by podhrmic