mavlink/rust-mavlink #72: Idiomatic way of accessing parsed message fields?

quietlychris, 2020-08-14:
Hello,

I'm working on using some of the data from a prototype [Aqualink ASV](https://discuss.bluerobotics.com/t/full-assembly-and-build-instructions-now-available-for-asv/7849), and am curious if I'm missing the idiomatic way of directly parsing and accessing a data structure read from a Mavlink message. 

I'm using something like the following to collect the messages:
```rust
let mut msgs: Vec<mavlink::common::MavMessage> = Vec::new();
loop {
    match vehicle.recv() {
            Ok((_header, msg)) => {
                println!("msg: {:?}\n",msg);
                msgs.push(msg);
            }
            Err(e) => {
                println!("e: {:?}",e);
                match e {
                    MessageReadError => {
                        //no messages currently available to receive -- wait a while
                        thread::sleep(Duration::from_millis(10));
                        continue;
                    }
                    _ => {
                        println!("recv error: {:?}", e);
                        break;
                    }
                }
            }
        }

        // Parse the last message in the queue
        let last_msg = &msgs.last().cloned().expect("Couldn't get the last message");
        let id = mavlink::common::MavMessage::message_id(last_msg);
        let payload = mavlink::common::MavMessage::ser(last_msg);

        let data = mavlink::common::MavMessage::parse(
            mavlink::MavlinkVersion::V2,
            last_msg.message_id(),
            &last_msg.ser(),
        ).expect("Couldn't parse data");

    // For example
    match id {
             30 => {
                 let attitude: Attitude = parse_attitude_message(last_msg).expect("Couldn't parse attitude message");
                 println!("Attitude: {:#?}", data);
             }
            _ => continue

}
```
The problem is, I'm taking the data and parsing it into my custom `Attitude` structure, by doing something like this: 
```rust
#[derive(Debug, Clone, Copy)]
pub struct Attitude {
    pub roll: f32,
    pub pitch: f32,
    pub yaw: f32,
    pub rollspeed: f32,
    pub pitchspeed: f32,
    pub yawspeed: f32,
}

pub fn parse_attitude_message_30(msg: &mavlink::common::MavMessage) -> Result<Attitude, Box<dyn Error>> {
    let data = mavlink::common::MavMessage::parse(
        mavlink::MavlinkVersion::V2,
        msg.message_id(),
        &msg.ser(),
    )?;
    
    // It seems like there should be a function somewhere that extends this match statement out and returns 
    // the information in a mavlink::common::INFO_DATA struct right back. 
    let attitude = match data {
        mavlink::common::MavMessage::ATTITUDE(mavlink::common::ATTITUDE_DATA {
            time_boot_ms: _,
            roll,
            pitch,
            yaw,
            rollspeed,
            pitchspeed,
            yawspeed,
        }) => Attitude::new(
            roll,
            pitch,
            yaw * 180.0 / PI, // yaw is now going to be in degrees instead of radians
            rollspeed,          // However, the "speed" variables should stil be in rad/s
            pitchspeed,
            yawspeed * 180.0 / PI, //
        ),
        _ => panic!("Error while parsing message") // If it's not an Attitude message, then we need to break off
    };
    Ok(attitude)

}
```

Honestly, this seems a little silly; I know that there's a `mavlink::common::ATTITUDE_DATA` data structure with named fields *somewhere*, but I'm having an issue with directly instantiating it so I can directly access the data using something like
```rust
let attitude: mavlink::common::ATTITUDE_DATA = {/* something */};
println!("The roll value is {}", attitude.roll); 
```
Of course, this isn't too much of an issue with a single struct, but if I want to access multiple fields at once (say, for example, I want Attitude and GPS (through the `mavlink::common::MavMessage::GLOBAL_POSITION_INT`, I now need to write another ~50 LoC of boilerplate to have access to those values. 

So, I guess my question is: am I missing something? I've spent a fair amount of time reading through the docs, and I know there's a `parse()` function that keeps showing up, but I'm having trouble figuring out how do combine all of these things in a better-looking way that still gets me what I'm looking for.

patrickelectric, 2020-08-15:
Hi @quietlychris, I would recommend to take a look in the [projects](https://github.com/mavlink/rust-mavlink#community-projects) that are using rust-mavlink as a guide of how to use the API.

But to answer your questions, this is how you can grab and fetch the message information:
```rust
let mut msgs: Vec<mavlink::common::MavMessage> = Vec::new();
loop {
    match vehicle.recv() {
            Ok((_header, msg)) => {
                println!("msg: {:?}\n",msg);
                match last_msg {
                    mavlink::common::MavMessage::ATTITUDE(attitude) => {
                        println!("Attitude: {:#?}", attitude);
                    }
                }
            }
            Err(e) => {
                println!("e: {:?}",e);
                match e {
                    MessageReadError => {
                        //no messages currently available to receive -- wait a while
                        thread::sleep(Duration::from_millis(10));
                        continue;
                    }
                    _ => {
                        println!("recv error: {:?}", e);
                        break;
                    }
                }
            }
        }
}
```

And after that you can do:
```rust
MyAttitude::new(
    attitude.roll,
    ... // and all the fields of your personalized struct
),
```

quietlychris, 2020-08-15:
Ah, perfect. I see, it looks like I was just passing the `mavlink::common::MavMessage::ATTITUDE` back into an identical, but temporarily-allocated structure that passed out of scope, rather than actually assigning it to a variable like I should have done.

 Sorry about that, now that you've pointed it out, I'm seeing this Waldo in those examples everywhere. Thanks :)

closed 2020-08-15 by quietlychris

patrickelectric, 2020-08-15:
@quietlychris be free to share with us your projects in our README that is using rust-mavlink. 
Have a great weekend!