mavlink/rust-mavlink #82: Don't support signing?

hamishwillee, 2021-01-27:
Just confirming for the mavlink docs!

Are there any more docs than provided on the readme? 
For context, I'm trying to characterize the things provided by each project
- ie level of docs (where [this one for C](https://mavlink.io/en/mavgen_c/) might be full
- level of testing - ie what is tested, unit test, round trip tests, test that send receive from C library etc.

patrickelectric, 2021-01-28:
Hi @hamishwillee, I hope that you are doing fine.

About rust-mavlink, it indeed has a support for signed messages, but not fully.
It can receive but will do nothing with the signed value itself.

For documentation, you can get it here: https://docs.rs/mavlink/latest/mavlink/
For testing: https://github.com/mavlink/rust-mavlink/tree/master/tests
    - They are done with some hardcoded messages to check if the pack/unpack values are correct and a self test of the internal functions.

hamishwillee, 2021-02-02:
Thank you very much for the clarification.

hamishwillee, 2021-02-02:
Note, I've updated mavlink docs with link in https://github.com/mavlink/mavlink-devguide/pull/312

I have not updated the information on signing because it still isn't quite clear to me what you mean by not supported. Full signing support would look something like:
1. Has a mechanism for setting a key to use for signing messages (key management itself might be in hardware).
1. Has mechanism for turning signing on and off
1. If signing turned on will sign MAVLink 2 messages.
1. If signing not turned on will parse signed and unsigned messages
1. If signing turned on will reject unsigned messages by default, but has hooks to allow an implementation to choose to keep some messages.

Assuming we're aligned on the above can you confirm again which it does/does not do?

tjrmarques, 2025-03-12:
Hi!
It's been a while since this has been merged, any idea on when it will land in crates.io?

patrickelectric, 2025-03-12:
@tjrmarques it appears that the release process has changed on rust-mavlink and more work on that is now necessary to do the release, I'm out of time at the moment.

pv42, 2025-05-23:
@hamishwillee the newest release now supports signing to the following extend:

1. Has a mechanism for setting a key to use for signing messages (key management itself might be in hardware).
Yes

2. Has mechanism for turning signing on and off
Yes

3. If signing turned on will sign MAVLink 2 messages.
Yes (can be disabled to only receive signed messages send unsigned ones)

4. If signing not turned on will parse signed and unsigned messages
Yes

5. a) If signing turned on will reject unsigned messages by default,
 Yes
b) but has hooks to allow an implementation to choose to keep some messages.
No (It is possible to allow all unsigned messages but not specific ones)

onur-ozkan, 2025-06-21:
We support signing, so I am closing this. For the missing features, we can create a tracking separate issue.

closed 2025-06-21 by onur-ozkan