mavlink/rust-mavlink #85: Add deployment to crates.ok

mrpollo, 2021-03-06:
It would be awesome if we could have the package in crates.io

This was originally brought up by @patrickelectric on the PX4 Slack #mavlink channel

This is an isssue to track.

patrickelectric, 2021-03-06:
Hi @mrpollo, sorry if I was not clear. We have the package under https://crates.io/crates/mavlink, in the pass it was done manually, after some work we put that to be automatic done by travis, and not we moved from travis to github actions, and now we need to add the secret key on the repository to be able to do the automatic package update on crates.io. Since I don't have permission to do such thing it's necessary to someone that have such permission to add the key.

kevinmehall, 2021-03-07:
@podhrmic set this up before. There's a [rust-mavlink-builder](https://crates.io/users/rust-mavlink-builder) account on GitHub / crates.io whose [crates.io API key was added as a secret in the Travis config](https://github.com/mavlink/rust-mavlink/commit/c975d078e5444114f3f608dc0cb72e2c177d8b75#diff-6ac3f79fc25d95cd1e3d51da53a4b21b939437392578a35ae8cd6d5366ca5485) for automatic publishing.

podhrmic, 2021-03-08:
Correct. @kevinmehall do you need me to change anything about the builder account, or can you just copy the API key to github actions?

patrickelectric, 2021-03-08:
I believe that only the API key on the [repository secrets](https://github.com/mavlink/rust-mavlink/settings/secrets/actions) should be enough

mrpollo, 2021-03-08:
@podhrmic @kevinmehall can the secret be copy-pasted? or should we re-generate the secret?

I went ahead with the copy/paste route and added the secret, as a "Repository Secret"

[image]

kevinmehall, 2021-03-08:
The value is encrypted in the .travis.yml and I don't think Travis offers a way to get it in cleartext, so @podhrmic will need to generate a new token and add it on GitHub's repository secrets.

kevinmehall, 2021-03-08:
I increased your permissions so you can get to the [repository secrets](https://github.com/mavlink/rust-mavlink/settings/secrets/actions)

podhrmic, 2021-03-08:
Token updated. It should be working now. How can we test it?

patrickelectric, 2021-03-08:
I'll soon create the build step for crates.io deployment

patrickelectric, 2021-03-08:
Done: https://github.com/mavlink/rust-mavlink/pull/92

mrpollo, 2021-03-09:
Thanks everyone 👏

closed 2021-03-09 by mrpollo