mavlink/rust-mavlink #93: New 0.9.0 version

patrickelectric, 2021-03-09:
I'm planning to deploy this week the 0.9.0 version with some new features including the new embedded support.
I'll just finish a couple of tests with the stm32f303 before doing it.
@podhrmic @kevinmehall let me know if you guys are ok with it

patrickelectric, 2021-03-11:
The new release will be done after https://github.com/mavlink/rust-mavlink/pull/94 get merged

patrickelectric, 2021-03-12:
0.9 is available, thanks for everybody that make it possible!

closed 2021-03-12 by patrickelectric

podhrmic, 2021-03-12:
Good work!

I noticed a funny thing - the README on [crates.io](https://crates.io/crates/mavlink) still shows version 0.8, even though the install instructions and everything else is up to date and shows version 0.9 - not sure how that happened, since the README is a mirror of the github repo...