mavlink/rust-mavlink #97: component_id does not support mavlink::common::MavComponent

patrickelectric, 2021-05-20:
MavComponent should be allowed to be used with component_id

hamishwillee, 2021-05-26:
@patrickelectric What do you mean? There is nothing called MavComponent in common.xml. 

Do you mean [MAV_COMPONENT](https://mavlink.io/en/messages/common.html#MAV_COMPONENT). If so this is an enum of component ids that have some predefined meaning. In theory you can actually any id other than 0 for anything, but in practice this will break code that relies on the predefined meaning.

patrickelectric, 2021-05-26:
Thanks for your input @hamishwillee ! But yes, this is more related to the rust implementation :smile: 
Sorry for creating this simple issue, but it's better to write it down than forgetting. 
The idea is to allow component_id used for the header creation to accept MavComponent enum, right now it only supports unsigned 8bits:
https://docs.rs/mavlink/0.9.0/mavlink/struct.MavHeader.html

But it should also accept MavComponent type:
https://docs.rs/mavlink/0.9.0/mavlink/common/enum.MavComponent.html

Probably I'm going to update the enum to accept an unspecified ID.
`MavComponent::UNSPECIFIED(id_that_does_not_exist_in_enum)`

hamishwillee, 2021-05-27:
No need to be sorry - I think it is excellent to create issues as it is way too easy to lose track of things! And this is definitely a bug.