mavlink/rust-mavlink #108: Serde derive macro is not found

elpiel, 2022-02-03:
I'm using mavlink with the default set of features and I get an error message for serde derive macro:

```
error: cannot find derive macro `Deserialize` in this scope
    --> /mnt/Disk/Projects/Personal/realsense/target/debug/build/mavlink-5f7d13b37869c4dd/out/common.rs:1144:49
     |
1144 | #[cfg_attr(feature = "serde", derive(Serialize, Deserialize))]
     |                                                 ^^^^^^^^^^^
     |
note: `Deserialize` is imported here, but it is only a trait, without a derive macro
    --> /mnt/Disk/Projects/Personal/realsense/target/debug/build/mavlink-5f7d13b37869c4dd/out/common.rs:23:13
     |
23   | use serde::{Deserialize, Serialize};

```

patrickelectric, 2022-02-03:
Hi @elpiel, I did a full cargo clean and cargo build, I'm using the latest stable version of rust: 1.58.1 (db9d1b20b 2022-01-20)

elpiel, 2022-02-03:
I believe that the serde feature should also enable `serde/derive`?
Once I added serde with the derive features to my Cargo.toml it compiles successfully

elpiel, 2022-02-03:
- Cargo.toml
```
[package]
name = "test-mavlink"
version = "0.1.0"
edition = "2021"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]

mavlink = "0.10"
```

- main.rs

```
fn main() {
    println!("Hello, world!");
}
```

- Run: `cargo build`

patrickelectric, 2022-02-03:
Thnaks for the example, indeed as a dependency the package is missing the derive feature.
I have created a PR to fix the issue: #109
You can test with: `mavlink = { git = "https://github.com/patrickelectric/rust-mavlink", branch = "fix_serde" }`
A 0.10.1 should be available soon

closed 2022-02-03 by patrickelectric

patrickelectric, 2022-02-03:
0.10.1 should be available now

PR #109 (2022-02-03): Cargo: Add serde derive on dependencies

Fix #108