mavlink/rust-mavlink #120: TuneFormat does not exist on rust-mavlink

patrickelectric, 2022-08-17:
https://mavlink.io/en/messages/common.html#TUNE_FORMAT

Erk-, 2022-08-22:
This will be there if the `mavlink` submodule is updated:
[image]

patrickelectric, 2022-08-22:
> This will be there if the `mavlink` submodule is updated: [image]

master branch of rust-mavlink does not build with the latest version of the mavlink repository. 

can you specify the mavlink hash that builds and contains tune format ?

patrickelectric, 2022-08-22:
Well, it appears that is showing now, closing it.

closed 2022-08-22 by patrickelectric