mavlink/rust-mavlink #135: Rust enums don't support combining, can't set multiple bitflags

kidevuojarvi, 2022-11-14:
Some of the bitflag data in the project is stored with enums, which causes issues with flags that are not mutually exclusive. Rust doesn't support combining enumerated values, and thus prevents the use of multiple flags, even though the flags are cast into `u16` in serialisation. 

The specific case I ran into is `mavlink::common::GimbalDeviceFlags`, where I'd like to lock each axis. This requires setting three flags, which can be done with bitwise-OR in C++, but is currently impossible with the Rust implementation.

Erk-, 2022-11-14:
It seems the reason that it fails is because it is not listed with `display = "bitmask"` in the `<field>`events which is what is used to decide if it should be a bitflag or a enum. This could be resolved by doing it a different way or doing some patching of the files when the code is generated.

kidevuojarvi, 2022-11-14:
Ah, apparently. Thanks for the quick reply, I'll look for a workaround.

Erk-, 2022-11-14:
If you make a folder called `patches` in the `build` folder you can drop patches that will be applied when compiling there