mavlink/rust-mavlink #190: Missing fields in some ardupilotmega structs

ExpertOfNil, 2023-08-10:
Cargo.toml entry:
```
[dependencies]
mavlink = "0.11.2"
```

I've run into an issue, when composing certain structs, where the documentation (and linked source) shows a public field is present, but the field is not recognized locally.  Here are a couple examples:
```rust
use mavlink::ardupilotmega::{MavMessage, SET_GPS_GLOBAL_ORIGIN_DATA};
let message = MavMessage::SET_GPS_GLOBAL_ORIGIN(SET_GPS_GLOBAL_ORIGIN_DATA {
    target_system: self.ctx.config.drone.system_id,
    latitude: req.latitude,
    longitude: req.longitude,
    altitude: req.altitude,
    time_usec: 0_u64, // no such field
});
```
```rust
use mavlink::ardupilotmega::{MavMessage, ATT_POS_MOCAP_DATA};
let message = MavMessage::ATT_POS_MOCAP(ATT_POS_MOCAP_DATA {
    time_usec: 0,
    q: [1_f32, 0_f32, 0_f32, 0_f32],
    x: 0_f32,
    y: 0_f32,
    z: 0_f32,
    covariance: [0_f32; 21],  // no such field
});
```
These are the two I can recall off-hand.  Any clarity here would be appreciated.

nknotts, 2023-08-12:
The missing fields are marked as extensions in the message definition xml. To enable extensions, you must add the `emit-extensions` feature. Example: add to `Cargo.toml`

```
mavlink = { version = "0.11.2", features = ["emit-extensions"] }
```

closed 2023-08-12 by patrickelectric

ExpertOfNil, 2023-09-21:
> The missing fields are marked as extensions in the message definition xml. To enable extensions, you must add the `emit-extensions` feature. Example: add to `Cargo.toml`
> 
> ```
> mavlink = { version = "0.11.2", features = ["emit-extensions"] }
> ```

Thank you for clearing that up for me!