mavlink/rust-mavlink #203: Make raw messages more flexible

brkydnc, 2023-11-07:
Currently, the `MAVLinkV1MessageRaw` and the `MAVLinkV2MessageRaw` types prevent mutable access to the internal buffers they use to store the raw bytes coming from any type of connection, which is problematic for libraries that built on top of this crate. 

What is actually being problematic is that this crate *enforces* you to use the *synchronous* reads (e.g `read_v2_raw_message`) in order to *fill* the raw message buffer *using* a reader, and there is no other way.

Building functionality over these raw packets is *exclusive* to the `rust-mavlink` library at the moment, preventing other libraries to interact with `rust-mavlink` in a flexible way. For example, to be able to implement a new `read()` function (e.g an async) for MAVLink messages from *another* library, the internal access to the internal buffer of a raw message is required. which this is not possible at the moment.

I highly suggest that these types *somehow* allow mutable access to the internal buffers.

patrickelectric, 2023-11-08:
Hi @brkydnc, sadly I'm out of time to help improve the library. But I'm here to help if you are willing to change what is necessary.

patrickelectric, 2023-11-08:
Probably related to: https://github.com/mavlink/rust-mavlink/issues/95

PR #382 (2025-08-26): Expose inner buffer
This allows external parsers to be used with this library.

Close #203
closed 2025-08-26 by onur-ozkan