mavlink/rust-mavlink #211: All enum entries and message ids use isize, some should use usize

antonio-sc66, 2023-12-12:
By default, Rust uses isize for the internal representation of enumerate fields. Negative numbers don't make sense for the [_enum entry_ fields](https://mavlink.io/en/guide/define_xml_element.html#enums) of the commands.

Example:

```
<entry value="32000" name="MAV_CMD_CAN_FORWARD" hasLocation="false" isDestination="false">
  <description>Request forwarding of CAN packets from the given CAN bus to this component. CAN Frames are sent using CAN_FRAME and CANFD_FRAME messages</description>
  <param index="1" label="bus">Bus number (0 to disable forwarding, 1 for first bus, 2 for 2nd bus, 3 for 3rd bus).</param>
  <param index="2">Empty.</param>
  <param index="3">Empty.</param>
  <param index="4">Empty.</param>
  <param index="5">Empty.</param>
  <param index="6">Empty.</param>
  <param index="7">Empty.</param>
</entry>
```

The same applies for message ids. Example:

```
<message id="12920" name="HYGROMETER_SENSOR">
  <description>Temperature and humidity from hygrometer.</description>
  <field type="uint8_t" name="id" instance="true">Hygrometer ID</field>
  <field type="int16_t" name="temperature" units="cdegC">Temperature</field>
  <field type="uint16_t" name="humidity" units="c%">Humidity</field>
</message>
```

Message ids and MAV_CMD entries always use positive numbers, and I don't see a reason why anybody would want them to be negative (in normal enums it does make sense to have entries with negative numbers)

The internal representation of usize and isize [depends on the arch type](https://doc.rust-lang.org/src/core/num/mod.rs.html#405)

Generated enums should force the use of usize. That behaviour can be changed as indicated in this [link](https://doc.rust-lang.org/reference/items/enumerations.html#assigning-discriminant-values).

In small embedded devices this change makes an important difference, as this kind of errors occur.

Example from an Atmega2560 from an Arduino Mega board.

```
target\avr-atmega2560\debug\build\mavlink-90379388018bf1fb\out\common.rs:1:95812
  |
1 | ... 10001 , MAV_CMD_PAYLOAD_PREPARE_DEPLOY = 30001 , MAV_CMD_PAYLOAD_CONTROL_DEPLOY = 30002 , MAV_CMD_FIXED_MAG_CAL_YAW = 42006 , MAV_CMD_DO_WINCH = 42600 , 
                                                                                                                                                         ^^^^^
  |
  = note: the literal `42600` does not fit into the type `isize` whose range is `-32768..=32767`
```

Required solution
----------------------
- pub enum MavMessage --> Should be generated using usize instead of the default isize
- MAV_CMD entries --> Should be generated using usize instead of the default isize

patrickelectric, 2023-12-12:
Great find!

antonio-sc66, 2023-12-12:
I've found more information.

Message ids
----------------
From [message schema](https://mavlink.io/en/guide/xml_schema.html#messages):
```
message: ...
  id: ...
    ...
    For MAVLink 2: Valid numbers range from 0 to 16777215. ...
```

So **u32** should be used **instead of isize** for the **MavMessage** enum definition

Normal enums
------------------
In terms of **normal enums**. I've seen that values up to i32::MAX are being used.

Example:
```
<enum name="MAV_SYS_STATUS_SENSOR" bitmask="true">
      ...
      <entry value="2147483648" name="MAV_SYS_STATUS_EXTENSION_USED">
        <description>0x80000000 Extended bit-field are used for further sensor status bits (needs to be set in onboard_control_sensors_present only)</description>
      </entry>
```

As differentiating between normal enums and the MAV_CMD enum might be cumbersome, and because it's usually better to have fixed size references instead of having different ranges in different devices maybe, the following patter should be used:

Proposed solution 2
-----------------------
- pub enum MavMessage -> u32
- enum entries representation -> i32

PR #214 (2024-01-03): Bugfix. Replace default isize enum representation.
By default Rust uses isize to represent enum values. isize implementation depends on the `target_pointer_width` attribute, which differs from arch to arch.

- Mavlink implementation for v2 requires message id with the range 0..16777215 --> u32
- Mavlink implementation requires u32 enum entry values

Closes #211
closed 2024-01-03 by patrickelectric