mavlink/rust-mavlink #212: 0.12.2 does not compile

gwbres, 2023-12-18:
Hello,

I'm interested in building custom Mavlink protos. Therefore I download and try to build the simplest form of this repository:

```bash
cargo build --features common
```

But this fails due to errors in the generated code related to Bitflags types.

[image]

A bitflag can only be dumped using `BitFlags::to_bits()`.

And the mirror operation :

[image]

A bitflag is not a primitive type. It should be built with `BitFlags::from_bits(u32)`

It seems that your build script already contains an attempt at dealing with those cases:

[image]

but for some reason we windup in the other branch. Unfortunately this is very cryptic to me and am unable to provide help.

* Am I missing something when building your library ?
* obviously 0.12.0 was released for a reason, why am I unable to build this "simple" form ?
* I'm also surprized you seem to get `uavionix` protocol, event hough you only asked for `common`. Which, in my case triggers more problem and gets more tedious. I my personcal case (only interested in my custom protocol), I would rather only activate `common` and my custom protocol, get rid of all the others

patrickelectric, 2023-12-18:
Hi @gwbres,

I would recommend to update your rust version with `rustup update` and `rustup override set stable`. To test I did build it with stable and nightly as well.

[image]

Try also a cargo clean and a cargo update just to be sure.
You can also check the steps on our ci that it's also passing with all tests.
How did you check for uavionix ? Only common should be integrated if you choose such feature.

gwbres, 2023-12-19:
@patrickelectric, 

first of all, thank you for your support ! 
I confirm that after a cargo clean, rustup update & rebuild it did pass.  
It was either a git cache or a cargo cache "issue", but that is not clear to me.

closed 2023-12-19 by gwbres