mavlink/rust-mavlink #224: MAVLink XML change to allow exponentiation operator in bitmask flag

hamishwillee, 2024-02-15:
FYI We're modifying the MAVLink XSD validation file in https://github.com/ArduPilot/pymavlink/pull/920 to allow a bitmask value to be declared using the Python exponentiation operator. 

This makes it easier to immediately visualize which bit is set by a flag, because you can use the syntax 
```xml
<entry value="2**15" name="BIT15" />
```
rather than
```xml
<entry value="32768" name=" BIT15" />
```
This is transparent to mavgen because the parser evaluates the number before creating the generated headers. However it may affect other the rust parser
We're not updating the XML yet to allow other parsers time to update.

Any concerns, please raise on https://github.com/ArduPilot/pymavlink/pull/920

patrickelectric, 2024-02-27:
Thanks for letting us know @hamishwillee

hamishwillee, 2024-02-28:
No worries. The change went in, but we haven't updated any of the messages yet. Waiting a few weeks to give people some catchup time.

closed 2026-02-12 by onur-ozkan