mavlink/rust-mavlink #227: Broken `no_std` from recent commits

peterkrull, 2024-03-05:
The commits relating to the new `PeekReader`, and some changes to the error types (specifically the use of std::io::Error) from 7d6a1c4 have made the library not usable in `no_std` environments.

tjrmarques, 2024-03-20:
Hi @peterkrull ,

I may have a fix in my fork https://github.com/tjrmarques/rust-mavlink, in the `backtracking-parser-nostd` branch.

Can you please check if it solves the issue?
If it does, I'll post a pull request.
If it doesn't, can you please provide a sample project/target that fails, as I don't have a no_std environment?

peterkrull, 2024-03-28:
Thanks for taking a look at it. I am not able to test it right now, since I use a branch with some modifications on top of it in my project. But I think you can test compatibility yourself by just importing the library in a bin crate with `#![no_std]` in the main file. If the compiler does not freak out, it is likely no_std compatible. The Github actions also fail on the current version, so that can also be used as an indicator.

tjrmarques, 2024-03-28:
No errors on my side, some tests do complain about embedded_hal on my machine, which I guess is fine.
CI seems happy though!

PR #233 (2024-03-28): Simplified PeekReader for MAVLink and made it `no_std` friendly
This should fix #227
closed 2024-03-28 by patrickelectric