mavlink/rust-mavlink #231: Incremental build broken (mavlink always regenerated/recompiled)

tjrmarques, 2024-03-20:
Since we moved to the new folder structure (09411e9ae8e26aef0d5b576e4a180ae3819cd13f) incremental compiles are broken.
In particular, the mavlink build script and corresponding code compilation are always run, regardless of whether or not there were changes.
This is especially problematic for projects that take `rust-mavlink` as a dependency, as build times will always be very long for every tiny change they make on their own code.

qwerty19106, 2024-03-22:
This is very annoying, I was forced to roll back to 12.2.
@GrahamDennis You can fix it?

JorgeCastanho, 2024-03-26:
I was also having this problem and found out it might be because now on `mavlink-bindgen` `cargo:rerun-if-changed` is only being checked in the end of `build.rs` file. I changed it to be done at the end of each iteration in the `_generate()` function's loop and is working as expected.
I can submit a PR if needed.

patrickelectric, 2024-03-28:
I believe that this is fixed by #232

closed 2024-03-28 by patrickelectric