mavlink/rust-mavlink #256: Async Connection Trait

pv42, 2024-08-22:
With #255 support for `tokio`-based `async` was merged, adding `async` versions of `write_versioned_msg` and `read_v2_msg` which is very welcome (as it is something I was thinking of doing myself).

I suggest adding an `async version of the connection trait similar to `MavConnection<M: Message>`: 
```rs
#[async_trait]
pub trait AsyncMavConnection<M: Message> {
    async fn recv(..) -> ..
    async fn send(..) -> ..
    fn set_protocol_version(..)
    fn get_protocol_version(..) -> ..
    async fn send_frame(..) -> ..
    async fn recv_frame(..) -> .. 
    async fn send_default(..) -> ..
}
```
And then implementing it for the `tcp`, `udp`, `file` and potentially `serial` (unsure if/how `tokio` support that) connections and providing a public function to open `async` connections:
`pub async fn connect_async<M: Message>(address: &str) -> io::Result<Box<dyn AsyncMavConnection<M> + Sync + Send>>`

If this is something that would fit and if @joaoantoniocardoso is not already working on it I would be happy to submit a PR.

joaoantoniocardoso, 2024-08-22:
Hi @pv42, nice idea, surely [mavlink-camera-manager](https://github.com/mavlink/mavlink-camera-manager) (and potentially [mavlink2rest](https://github.com/mavlink/mavlink2rest)) would benefit from that!

To contextualize, I did implement these tokio-based async read/write because I've been experimenting with a new mavlink router written in rust ([mavlink-server](https://github.com/bluerobotics/mavlink-server)). This router can also be refactored to make use of that, but it already has an implementation (which might help us here?).

Feel free to use any of that code if useful, the mavlink-server repo still lacks a license file, but it's intended to be MIT.

BWStearns, 2024-08-22:
I was literally looking to see if this existed last night! +100!

BWStearns, 2024-08-23:
Given #255 what does an async connection trait look like broad strokes? Like what would a ticket look like for this?

joaoantoniocardoso, 2024-08-23:
@pv42 just so you know, I'm reworking the tokio async reader around tokio-utils' codec, so we can ensure a frame would be received from the raw read functions, and using the same codec you can have a tokio's UdpStream.

closed 2024-09-17 by pv42