mavlink/rust-mavlink #258: MSRV for tokio-1 feature

pv42, 2024-08-22:
I noticed that the `tokio-1` feature depends on `tokio^1.0` which potentially violates the MSRV check since the latest `tokio` 1.39 requires Rust 1.70. This was likely not caught since the test does not enable the feature for it. The problem can be fixed by either:
- restricting tokio to <= 1.38 (MSRV 1.63)
- buming this crates MSRV to 1.70; since this crate does not appear to have an exception specified that would be semver breaking
- specify a seperate MSRV policy for `tokio-1`

See [tokio MSRV](https://github.com/tokio-rs/tokio?tab=readme-ov-file#supported-rust-versions)

joaoantoniocardoso, 2024-08-24:
Hi, I just noticed that I mistyped the version I wanted to use when setting the `tokio-1` feature: it was supposed to be `^=1`, not `^=1.0`. :sweat_smile: 

Personally, all the projects I am working on use the latest rust version, so I'd prefer either bumping this crate MSRV to `1.70` (and using tokio `^=1`) or specifying a separate MSRV for `tokio-1`.

I git-blamed to track the MSRV changes:
- https://github.com/mavlink/rust-mavlink/commit/09411e9ae8e26aef0d5b576e4a180ae3819cd13f
- https://github.com/mavlink/rust-mavlink/commit/74243e339a131d8ca08e26f51b900f0d373eae10
- https://github.com/mavlink/rust-mavlink/commit/c0417b5cf20d0edd418d2f018163eccb1b8e9deb
- https://github.com/mavlink/rust-mavlink/commit/d1caf7a78fe4e6d98ac04ded5cd5d50dbe29ad02

The people responsible for those changes might want to participate here as well: @GrahamDennis @danieleades @patrickelectric 

thanks!

danieleades, 2024-08-25:
i don't have a strong preference what the solution is, so long as the MSRV is comprehensively checked so that it changes intentionally and predictably. Clearly there's a gap now in that the features are not included in the MSRV check.
I think both solutions (having a separate MSRV for the options, or simply bumping the MSRV and updating the CI) are reasonable. I've got a preference for just having a single MSRV, for simplicity. But i'd be fine with either