mavlink/rust-mavlink #263: Better feedback on feature combinations

danieleades, 2024-08-25:
Would be great to have some documentation on valid combinations of features. This repo makes heavy use of features and some of these are mutually exclusive or have other relationships between them.

These should be documented and invalid combinations should have clear panic messages.

Would also be good to review that all of the valid combinations are tested in CI

danieleades, 2024-08-25:
Should also consider setting a custom default rust-analyzer command that doesn't emit every single target

pv42, 2024-08-25:
To also improve the documentation using `#![feature(doc_auto_cfg)]` to add those "Available on crate feature *featurename* only." badges would be a good idea. 
Apparently this is a nightly feature but docs.rs is on nightly so `#![cfg_attr(docsrs, feature(doc_auto_cfg))]` would work.

danieleades, 2024-08-25:
@patrickelectric can you provide any context on what combinations of features are allowed?

patrickelectric, 2024-08-25:
This was probably broken on the pass PRs, some work need to be done to fix that besides adding tests in CI.

danieleades, 2024-08-25:
> This was probably broken on the pass PRs, some work need to be done to fix that besides adding tests in CI.

I guess i'm trying to understand what is *expected* to work

patrickelectric, 2024-08-25:
> > This was probably broken on the pass PRs, some work need to be done to fix that besides adding tests in CI.
> 
> I guess i'm trying to understand what is _expected_ to work

At the moment, check the combinations in CI. For conflict configuration is still necessary to document.

PR #315 (2025-06-02): Documentation improvements
## Doc changes:
- Remove duplicate dots from generated code doc
- Url link in generated code will generate as links
- Fix missing links for io::ErrorKind
- A feature documentation to mavlink (this does not really resolve #263)
- capitalize MAVLink like this in documentaion
- MAVLink v1/2 ->  MAVLink 1/2
- remove duplicated comment about Vx maxium size
- read/write stream -> reader/writer
- add doc to most undocumented things
- remove old comments about dependent dialect features
- mention generated feature gates in generated code
- mention git dependency (see #306)
- [... cut]
closed 2025-06-02 by patrickelectric