mavlink/rust-mavlink #274: Fix release CI

patrickelectric, 2024-08-26:
At the moment release CI is broken, it appears that some necessary changes in cargo-workspaces is still not available to allow deployment of multiple crates in an workspace. It's necessary for us to come up with a solution meanwhile.

pv42, 2024-08-26:
As I understand the problem is that crate-interdepence via paths is not allowed on crates.io while using versions breaks tests that depend on changes ahead of the latest release. 
Is there anything preventing the use of both at the same time ?

patrickelectric, 2024-08-26:
> As I understand the problem is that crate-interdepence via paths is not allowed on crates.io while using versions breaks tests that depend on changes ahead of the latest release. Is there anything preventing the use of both at the same time ?

How do you suggest to have both simultaneously ? The release should be an automatic operation.

pv42, 2024-08-26:
Like this (this is a dev dependency but should be the same)
https://github.com/tokio-rs/tokio/blob/1ac8dff213937088616dc84de9adc92b4b68c49a/tokio-macros/Cargo.toml#L30
```toml
tokio = { version = "1.0.0", path = "../tokio", features = ["full"] }
```

pv42, 2024-08-26:
I don't think the resolver matters since the change seems to be mostly about features.
Anyways here is a workspace on resolver 1 (since it is edition 2018) that uses it: [hdf5](https://github.com/nlhepler/hdf5-rs/blob/master/Cargo.toml)

PR #301 (2025-05-22): Fix release CI
Attempts to fix #274.
- add version dependencies to for `mavlink-core` and `mavlink-bindgen to `mavlink`
- replace workspace publish with 3 single crate publishes, this is done since I could not get it to work with `--dry-run`, it is possible it works after all.
- The current version numbers on the different crates are kind of a mess but should be fixed by the commit created by the deploy pipeline depending on the latest tag (currently 0.13.2)
closed 2025-05-22 by patrickelectric