mavlink/rust-mavlink #306: Unable to build project on Ubuntu (error: failed to run custom build command for `mavlink v0.13.1 (https://github.com/mavlink/rust-mavlink.git?tag=0.13.2#c93fd578)`)

mclayton7, 2025-05-21:
I'm getting an error (see below) about running the custom build command for `mavlink v0.13.1`. Any ideas? I attached my `Cargo.toml` as well. Any ideas on why it's not building?

`cargo build` output:
```
root@b980996d6164:/src# cargo build
    Updating git repository `https://github.com/mavlink/rust-mavlink.git`
   Compiling proc-macro2 v1.0.95
   Compiling unicode-ident v1.0.18
   Compiling libc v0.2.172
   Compiling autocfg v1.4.0
   Compiling thiserror v1.0.69
   Compiling crc-any v2.5.0
   Compiling memchr v2.7.4
   Compiling serde v1.0.219
   Compiling parking_lot_core v0.9.10
   Compiling lazy_static v1.5.0
   Compiling syn v1.0.109
   Compiling smallvec v1.15.0
   Compiling scopeguard v1.2.0
   Compiling cfg-if v1.0.0
   Compiling byteorder v1.5.0
   Compiling lock_api v0.4.12
   Compiling num-traits v0.2.19
   Compiling bytes v1.10.1
   Compiling quick-xml v0.26.0
   Compiling bitflags v1.3.2
   Compiling pin-project-lite v0.2.16
   Compiling quote v1.0.40
   Compiling syn v2.0.101
   Compiling ioctl-rs v0.1.6
   Compiling termios v0.2.2
   Compiling serial-core v0.4.0
   Compiling mio v1.0.3
   Compiling serial-unix v0.4.0
   Compiling socket2 v0.5.9
   Compiling serial v0.4.0
   Compiling parking_lot v0.12.3
   Compiling signal-hook-registry v1.4.5
   Compiling num-derive v0.3.3
   Compiling thiserror-impl v1.0.69
   Compiling serde_derive v1.0.219
   Compiling tokio-macros v2.5.0
   Compiling tokio v1.45.0
   Compiling mavlink-bindgen v0.13.1
   Compiling mavlink v0.13.1 (https://github.com/mavlink/rust-mavlink.git?tag=0.13.2#c93fd578)
error: failed to run custom build command for `mavlink v0.13.1 (https://github.com/mavlink/rust-mavlink.git?tag=0.13.2#c93fd578)`

Caused by:
  process didn't exit successfully: `/src/target/debug/build/mavlink-d8eaf76fb728c3be/build-script-main` (exit status: 1)
  --- stderr
  No such file or directory (os error 2)
warning: build failed, waiting for other jobs to finish...
```

Cargo.toml:
```
[package]
name = "attnav_package"
version = "0.1.0"
edition = "2021"

[dependencies]
mavlink = { git = "https://github.com/mavlink/rust-mavlink.git", tag = "0.13.2" }
tokio = { version = "1.37.0", features = ["full"] }
```

rustup and rustc output:
```
root@b980996d6164:/src# rustup show
Default host: x86_64-unknown-linux-gnu
rustup home:  /root/.rustup

installed toolchains
--------------------
stable-x86_64-unknown-linux-gnu (active, default)

active toolchain
----------------
name: stable-x86_64-unknown-linux-gnu
active because: it's the default toolchain
installed targets:
  x86_64-unknown-linux-gnu
root@b980996d6164:/src# rustc --version
rustc 1.87.0 (17067e9ac 2025-05-09)
```

pv42, 2025-05-21:
Do you have `git` installed ?
From testing it seems not having that might cause this issue.

mclayton7, 2025-05-22:
@pv42 you are right that fixed it, thank-you so much!

closed 2025-05-22 by mclayton7