mavlink/rust-mavlink #311: 0.13.2 bindgen introduces arbitrary feature gate

Indeximal, 2025-05-25:
Combined with our `--Deny Warnings` lead our CI to spontaneous break. Adding feature gates to generated code doesn't seem particularly smart to me, since you can in/exclude it with options in the build script. Additionally, it now means the cargo toml of the crate using mavlink-bindgen has to be updated to make use of the new feature anyway, as the dependencies are not automatically used (e.g. by reexporting them in mavlink-bindgen).

Please be more careful with minor changes, a lot of things can be breaking.

joaoantoniocardoso, 2025-05-26:
I'm not sure if I understood the message. The 'arbitrary' feature should be on 0.14.0, not 0.13.2. But mainly, since there's no stable version (as in.. we are using `0.x.x`), a minor version will most certainly have breaking changes, otherwise it's set as a patch.

Indeximal, 2025-05-26:
> But mainly, since there's no stable version (as in.. we are using `0.x.x`)
This is not how [semver](https://semver.org) works. 0.x.y means that 0.x.z is comatible and 0.x+1.y is breaking.

> The 'arbitrary' feature should be on 0.14.0, not 0.13.2
The [crates.io for mavlink-bindgen](https://crates.io/crates/mavlink-bindgen/versions) clearly shows you published 0.13.2, no 0.14.0 in sight.

pv42, 2025-05-26:
There was a issue with the release and what should have been published as 0.14 didn't get properly version bumped.

joaoantoniocardoso, 2025-05-26:
> This is not how [semver](https://semver.org/) works. 0.x.y means that 0.x.z is comatible and 0.x+1.y is breaking.

Good, this was precisely what I meant by "minor" vs "patch".

joaoantoniocardoso, 2025-05-26:
>0.13.2 and 0.14.0 will be removed, the new version 0.14.1 should be used

Just to note, [the version `0.13.2` from crates.io](https://crates.io/crates/mavlink/0.13.2) is wrong, having the `arbitrary` feature. The [git tag `0.13.2`](https://github.com/mavlink/rust-mavlink/releases/tag/0.13.2) is correct.