mavlink/rust-mavlink #317: embedded example does not build

pv42, 2025-06-02:
The embedded example does not build when using the newest `0.7.5` version of `cortex-m-rt` (on native windows and wsl): 
```
   Compiling mavlink-embedded v0.1.0 (/mnt/e/pv42/rust-mavlink/mavlink/examples/embedded)
error: linking with `rust-lld` failed: exit status: 1
  |
  = note:  "rust-lld" "-flavor" "gnu" "/tmp/rustcZan81K/symbols.o" [...]
  = note: some arguments are omitted. use `--verbose` to show all linker arguments
  = note: rust-lld: error:
          ERROR(cortex-m-rt): stack end address is not below stack start.

          rust-lld: error:
          ERROR(cortex-m-rt): stack end address is not below stack start.

          rust-lld: error:
          ERROR(cortex-m-rt): stack end address is not below stack start.
```
It works using the previous  `cortex-m-rt=0.7.3` (0.7.4 got yanked).

joaoantoniocardoso, 2025-07-03:
I'm only able to build it in the way that our CI is doing:

```bash
cargo build \
  --target thumbv7em-none-eabihf \
  --manifest-path mavlink/examples/embedded/Cargo.toml
```

pv42, 2025-07-03:
The only differnce is the folloing section in `mavlink/examples/embedded/.cargo/config.toml`:
```toml
[target.thumbv7em-none-eabihf]
rustflags = [
  "-C", "link-arg=-Tlink.x",
]
```
When it is removed the method described in the examples readme compiles.

The `cortex-m-rt` doc seems to suggest that this would be required, so maybe it breaks the resulting binary if omited:
https://docs.rust-embedded.org/cortex-m-quickstart/cortex_m_rt/index.html#an-example

joaoantoniocardoso, 2025-07-03:
@patrickelectric so.. maybe the lib _is_ broken for embedded.

pv42, 2025-07-03:
Yeah if STMCube generates that this it must be correct.

PR #346 (2025-07-16): Fix embedded example and ci and remove ardupilotmega feature
Closed #317

Fixes for embedded: 
- fix memory.x by adding `_stack_end`
- remove unused ardupilotmega dialect feature
- fix CI test not using linker script `link.x`