mavlink/rust-mavlink #324: Offer to help maintain rust-mavlink

onur-ozkan, 2025-06-17:
We are working on low-powered systems and rely heavily on `rust-mavlink` (and really love it!). I sent couple of patches to upstream (here) and there will be even more in the future. I have been allocating time specifically to improve things  on rust-mavlink and send patches. However, progress on upstream seems a bit slow (issues aren't maintained well, project barely gets improved, some features are half-supported, there is only one reviewer (as far as I seen) and they aren't very active on rust-mavlink, etc.).

Totally understand if the current maintainers are already under a lot of pressure (if so, no worries at all, and thanks for all the work you have already done on the project). I initially posted about this in the Discord channel and one admin (Ramon) suggested opening an issue here to ask more formally.

We would prefer not to maintain a fork just for ourselves. If help is needed and there is room, I would be happy to contribute as an additional maintainer and reviewer to help keep the project on track.

patrickelectric, 2025-06-17:
That would be lovely! I just added you in the repository :) 
Sorry for not having a dedicated time to the project anymore

onur-ozkan, 2025-06-17:
Thank you :)

> Sorry for not having a dedicated time to the project anymore

No problem at all. Thank you for all the effort you have put in to bring the project this far!

patrickelectric, 2025-06-17:
If you can, please put myself and @joaoantoniocardoso to review the PRs, we'll try to review asap, if not, at least we are going to be sync to what is happening and collaborate when possible.

onur-ozkan, 2025-06-17:
I will add you (or at least ping you) on each PR when I target the master branch and a release. Same for major change proposals as well, I will share the design and the idea with you before going for the implementation. This weekend I am planning to prioritize some tasks and outline the current state of the project.

One thing I would like to do is creating custom label system (for each area and issue type) and organize issues on a project board (I maintain plenty of open source projects and visualizing priorities and project state helps me very often). But that requires to be a member of the org. Should I reach out to someone else about this (maybe Lorenz)?

hamishwillee, 2025-06-18:
@onur-ozkan If Patrick doesn't have bandwidth to set up the board for you then @mrpollo can, and I might be able to (I appear to have the magic rights, but not very familiar with that part of the infrastructure). All that needs IMO is broad agreement with @patrickelectric and @joaoantoniocardoso