mavlink/rust-mavlink #326: `MavConnection::recv` can cause `MavConnection::send` to hand

onur-ozkan, 2025-06-20:
We acquire the lock at the beginning of `MavConnection::send`. Which means, in the loop, if we repeatedly hit `_` part of the `match` (which is any error that isn't `ErrorKind::UnexpectedEof`) here
 https://github.com/mavlink/rust-mavlink/blob/7123184db4bcadb2f2ba88086fbf288a0d2ee1fe/mavlink-core/src/connection/direct_serial.rs#L41-L51

`MavConnection::send` will hang while trying to lock the port since `recv` never releases it. This happened to me couple times and I thought I have issues on my thread pool.

Ideally, we should return `io::ErrorKind::TimedOut` if no message is received within a certain time interval (e.g., 5 seconds).

Any opinions? @patrickelectric

onur-ozkan, 2025-06-20:
Alternatively we could separate the read-write locks so they don't block each other. Although I have to check if it's safe to do that.

pv42, 2025-06-20:
serialport-rs seem to support duplex by cloning the port object and then using one for reading and one for writing. I think this is a preferable approach since having the fixed read timeout would cause the send function to block until the read is timed out.

joaoantoniocardoso, 2025-06-20:
Reopening since the fix was reverted.

PR #329 (2025-06-20): refactor: split serial port into separate r/w mutexes
Use separate port instances for reading and writing operations to prevent deadlock when concurrent `recv()` and `send()` calls compete for the same mutex.

Resolves #326
PR #332 (2025-06-20): refactor: split serial port into separate r/w mutexes
Use separate port instances for reading and writing operations to prevent deadlock when concurrent `recv()` and `send()` calls compete for the same mutex.

Resolves #326
closed 2025-06-20 by onur-ozkan