mavlink/rust-mavlink #338: github pages documentation ignores `format-generated-code`

pv42, 2025-06-24:
The documentation generated https://mavlink.github.io/rust-mavlink/mavlink/ seems to ignore the format-generated code feature (see [minimal.rs](https://mavlink.github.io/rust-mavlink/src/mavlink/home/runner/work/rust-mavlink/rust-mavlink/target/debug/build/mavlink-f9215e42db300084/out/minimal.rs.html)). The CI command https://github.com/mavlink/rust-mavlink/blob/80e5747710a38fbbd3879f506067f3df0e952444/.github/workflows/test.yml#L130
should set the flag and it is used when used properly locally. All other feature flags also seem to work. 
The reason why this matters is that documentation with that flag is significantly slower (see #316). This significantly slows down the CI as a whole. The last PR docs run took 20 minutes. 
I am unsure why that is happening.

onur-ozkan, 2025-06-24:
Is there any specific reason why we generate this page?

pv42, 2025-08-26:
Caused by the CI env not having rustfmt installed.

PR #385 (2025-08-26): Fix doc CI formatting
Fixes #338
- The time of the docs Ci step went down from 21:09 in the last PR to 1:59 in this PR

The CI enviroment did not have rustfmt installed causing the `format_generated_code(..)` function to do nothing.
This PR:
- adds rustfmt to the CI
- document that rustfmt is required for this feature to work
- improve the error message, this will cause cargo to print a warning when running from a build script instead of using `eprintln!`. `eprintln!` is only visible with `-vv` and buried among a lot of other messages.

The warning message from the build script would look lith [... cut]
closed 2025-08-26 by patrickelectric