mavlink/rust-mavlink #347: limitations of dynamic connection dispatching

onur-ozkan, 2025-07-05:
I am working on adding read and write buffering for serial connections. But there is a problem: there is no clear way to handle connection-type specific configurations (e.g., adding options specifically for `SerialConnection` and exposing them to users). This is because the way connection is initialized is too generic and it's somewhat magically with hiding the actual connection type.

In action, this is what users do today to initialize connections and do some configuration:

```rust
let mut mavconn = mavlink::connect::<mavlink::ardupilotmega::MavMessage>(...);
mavconn.set_protocol_version(mavlink::MavlinkVersion::V1);
```

However, this approach can't work for every case. When connection parameters become configurable, setting them after initialization doesn't always make sense because the connection type is already settled.

Instead of this dynamic connection dispatch, I suggest moving to an explicit connection initialization approach. So then:

1. We don't have to adapt every connection type just because we want to add configuration for a specific one.
2. We can expose connection-specific configuration types that don't need to be generic or compatible across all connection types.

By roughly thinking, something like this would be simple and efficient:

```rust
// Using `SerialConnection` for demonstration purposes, this applies to each connection type.
let mut config = SerialConnectionConfig::new();
config.disable_x();
config.enable_y();
config.set_z(5);

let con = SerialConnection::connect(or new)(config);
```

cc @patrickelectric @joaoantoniocardoso @pv42

onur-ozkan, 2025-07-05:
> There are the already the implementations of the Connectable trait e.g. SerialConnectable. They currently just represent a parsed address string but adding more options there would be possible.

I am aware of that but it didn't look like a decent approach to add new fields there and then propagate them. To me, I think what we should do is just make the connection type more explicit like I proposed in the issue, or even something like this could help:

```rust
enum ConnectionConfiguration {
  Serial(SerialConnectionConfig),
  File(FileConnectionConfig),
  ...
}

// Update `mavlink::connect` function to take `ConnectionConfiguration`.
mavlink::connect(ConnectionConfiguration::Serial(config));
```

pv42, 2025-07-05:
I am personally in favor of at least having a `connect_str(addr: &str) -> ...` function exposed since it would allow to add the mavlink connection with only a single string parameter compared to the rather complex `ConnectionConfiguration`. This does not neccesary have to dyn it could also an `enum`. Alternatively at least something like `parse_address(str).connect()` should be present.

As for the ConnectionConfiguration enum proposed, I assume it would still contain the address in some form internally. So could you maybe clarify how it would be different from extending and renaming the `ConnectionAddress` enum.

onur-ozkan, 2025-07-07:
> I am personally in favor of at least having a `connect_str(addr: &str) -> ...` function exposed since it would allow to add the mavlink connection with only a single string parameter compared to the rather complex `ConnectionConfiguration`. This does not neccesary have to dyn it could also an `enum`. Alternatively at least something like `parse_address(str).connect()` should be possible.

I don't have any issues with keeping `connect($addr)` as it is, although I am not entirely convinced that passing a simple configuration is all that complex. For example if a user wants to rely on defaults, their usage might look like `connect(SomeConnectionConfig::new($addr))`.

> As for the ConnectionConfiguration enum proposed, I assume it would still contain the address in some form internally. So could you maybe clarify how it would be different from extending and renaming the `ConnectionAddress` enum.

The idea is to rework the `connectable` module into a config module and promote the configuration layer to a first-class citizen in the API. This would allow each connection protocol to expose its own specific configuration options.

joaoantoniocardoso, 2025-07-07:
That will be like a good improvement. I think we just need to make sure not to break how people have been using the library. For example, some of the applications I am aware of are directly passing the connection string from a CLI arg to the connect function. Just let's not make those applications' responsibility to parse the MAVLink connection strings into new connectable MAVLink objects.

I'm fine with both:
- `connect_str(&addr)`: requires an intuitive change;
- `connect(SomeConnectionConfig::new($addr))` seems like an easy change as well;

But what about:
- `connect(&addr)` + `connect_with(config)`: keep the current API intact, while still adding the config; This also allow us to expand the &addr to be a URI with the configs passed as query parameters, like `serial://<PORT>?baudrate=<BAUDRATE>&parity=<PARITY>` (we do like this in another project -- there are libs that can parse query parameters into structs, and then we use those structs as data for our builders, which is not hard to maintain);

thanks

pv42, 2025-07-08:
> For example, some of the applications I am aware of are directly passing the connection string from a CLI arg to the connect function.

This was my concern too, a way exists to still parse these connection strings should still exist but any of the ideas above would be ok for my use case. I think specifying a connection configuration should probably be the preferred  method of connecting and I would advocate for `connect` function, which the first thing users will try to use it. Since this proposal would change the API anyways I think this might be worth considering.

> The idea is to rework the connectable module into a config module and promote the configuration layer to a first-class citizen in the API. This would allow each connection protocol to expose its own specific configuration options.

I understand now what you meant, this is a great idea!

> This also allow us to expand the &addr to be a URI with the configs passed as query parameters, like serial://<PORT>?baudrate=<BAUDRATE>&parity=<PARITY> (we do like this in another project -- there [...]

My understanding was that the connection string tries to be somewhat compatible to what [pymavlink](https://mavlink.io/en/mavgen_python/#connection_string) is doing. Looking at it again there are some quite major differences (optimal protocol specifier etc.). 
I am uncertain if a URI parser should be part of mavlink itself or if users should implement it themselves when required, we could put it in an example. It would add a second way to connect via a string since I imagine we would have to keep supporting the old string for at least time.

onur-ozkan, 2025-08-02:
Sounds like the fundamentals of the idea are agreed upon. I will start working on the initial implementation and send a PR in a week or so (likely the next weekend). We can continue discussing the details directly on the code/PR, which should make the design process much easier than trying to do it here without a working code/example.

closed 2025-08-21 by onur-ozkan