mavlink/rust-mavlink #348: 0.15.0 bitflags serialization

cappuc, 2025-07-10:
Serialization of bitflags has changed from 0.14.1 to 0.15.0. Is this an intended change or a bug?

0.15.0 (Serialization of HEARTBEAT base_mode is a string: 
```
"MAV_MODE_FLAG_SAFETY_ARMED | MAV_MODE_FLAG_MANUAL_INPUT_ENABLED | MAV_MODE_FLAG_STABILIZE_ENABLED | MAV_MODE_FLAG_GUIDED_ENABLED | MAV_MODE_FLAG_CUSTOM_MODE_ENABLED"
```

0.14.1 (Serialization of HEARTBEAT base_mode is an object with bits as number:
```
{ bits: 217 }
```

Another thing I noticed: with  "format-generated-code" feature enabled the bitflags definition are not "pretty" formatted, for example:
```
bitflags! { # [cfg_attr (feature = "serde" , derive (Serialize , Deserialize))] # [cfg_attr (feature = "arbitrary" , derive (Arbitrary))] # [derive (Debug , Copy , Clone , PartialEq)] # [doc = "These flags encode the MAV mode."] pub struct MavModeFlag : u8 { # [doc = "0b10000000 MAV safety set to armed. Motors are enabled / running / can start. Ready to fly. Additional note: this flag is to be ignore when sent in the command MAV_CMD_DO_SET_MODE and MAV_CMD_COMPONENT_ARM_DISARM shall be used instead. The flag can still be used to report the armed state."] const MAV_MODE_FLAG_SAFETY_ARMED = 128 ; # [doc = "0b01000000 remote control input is enabled."] const MAV_MODE_FLAG_MANUAL_INPUT_ENABLED = 64 ; # [doc = "0b00100000 hardware in the loop simulation. All motors / actuators are blocked, but internal software is full operational."] const MAV_MODE_FLAG_HIL_ENABLED = 32 ; # [doc = "0b00010000 system stabilizes electronically its attitude (and optionally position). It needs however further control inputs to move around."] const MAV_MODE_FLAG_STABILIZE_ENABLED = 16 ; # [doc = "0b00001000 guided mode enabled, system flies waypoints / mission items."] const MAV_MODE_FLAG_GUIDED_ENABLED = 8 ; # [doc = "0b00000100 autonomous mode enabled, system finds its own goal positions. Guided flag can be set or not, depends on the actual implementation."] const MAV_MODE_FLAG_AUTO_ENABLED = 4 ; # [doc = "0b00000010 system has a test mode enabled. This flag is intended for temporary system tests and should not be used for stable implementations."] const MAV_MODE_FLAG_TEST_ENABLED = 2 ; # [doc = "0b00000001 Reserved for future use."] const MAV_MODE_FLAG_CUSTOM_MODE_ENABLED = 1 ; } }
```

joaoantoniocardoso, 2025-07-10:
Hi, thanks for reporting. We are clearly lacking some tests to keep the API safe -- this was _not_ intentional.

This was a change from the [bitflags 2.0.0 release](https://github.com/bitflags/bitflags/releases/tag/2.0.0):

> The default serialization format with serde has changed if you #[derive(Serialize, Deserialize)] on your generated flags types. It will now use a formatted string for human-readable formats and the underlying bits type for compact formats.
>
> To keep the old format, see the https://github.com/KodrAus/bitflags-serde-legacy library.

On our side, it was changed from `1.2.1` to `2.9.1` on [this PR](https://github.com/mavlink/rust-mavlink/pull/325).

patrickelectric, 2025-07-10:
to update on that, 0.10 is being maintained to avoid api breaks.